derkork/openscad-graph-editor #7: Not possible to drag/reposition nodes using a stylus

WillAdams, 2022-08-14:
If you have a Tablet PC running Windows, it is not possible to drag/reposition elements, or interact w/ nodes, instead one gets scrolling events:

[image]

derkork, 2022-08-14:
Hmm, i have tried it on my Windows PC with a Wacom tablet and this seems to work fine. I don't own a Windows tablet, so I currently have no idea how to reproduce/fix this.  Are there any settings in the windows tablet that allow you to configure how the stylus works (maybe even per app)?

WillAdams, 2022-08-15:
Not that really work --- the problem seems to be Microsoft Ink, and if I uninstalled it, I'd give up handwriting recognition --- there are some apps which afford some options, notably Google Chrome which is supposed to allow toggling on/off scroll behaviour, which I think is what is being expressed here.

derkork, 2022-08-15:
Hmm, I have tried both enabling and disabling Windows Ink and it doesn't seem to make a difference with my Wacom tablet.

derkork, 2022-08-22:
Do you have by any change press&hold for right-click enabled? https://support.wacom.com/hc/en-us/articles/1500006338262-The-pen-right-clicks-when-pressing-down-what-do-I-do-

WillAdams, 2022-08-22:
Yes, I do, since I use, by preference a Staedtler Noris Digital Stylus which lacks buttons entirely --- turning that off didn't change this though.

One interesting thing is that I can select an object and drag it w/ the stylus --- once --- but thereafter, trying to drag it doesn't work until it is deselected.

derkork, 2022-08-22:
I seem to learn a lot about styluses through this one  - one has to see the upsides. For debugging this I have added a small dialog to 0.3.1 where we can see the input events caused by the stylus. You can open it by pressing Ctrl+Shift+T. 

[image]

Then try to press and drag on the test area with the stylus. We should get `InputEventMouseButton` for clicks with the tip and `InputEventMouseMotion` when dragging over it. When I tried it with my Wacom I got a scroll event one time but only one time and I cannot for the life of me reproduce it. Could you give this a spin once I release 0.3.1 so we can maybe get to the bottom of this.

WillAdams, 2022-08-23:
Tried it, and the stylus events seem to be intercepted by the system before the program sees them.

I get:

 - DragUp
 - DragDown
 - Back
 - Forward

Checking everywhere for preferences:

the Samsung app only affords an option for the pen button (not applicable) and adjusting pressure sensitivity.

Windows 10 Pen Settings just has:

 - Show Visual Effect (on)
 - Show cursor (on)
 - Ignore touch input when using pen (enabled)
 - Show the handwriting panel when not in tablet mode and there's no keyboard attached (on)
 - Show recommended app suggestions (on)

and various Windows Ink Workspace settings which don't seem applicable

Checking in the older Windows Control Panel for Pen and Touch --- the problem seems to be "Flicks":

[image]

which once disabled, allowed dragging to work as expected.

(still not seeing stylus events that I can read in the nifty window though --- it changes from InputEventMouseMotion too quickly to something and back for me to read)

My apologies for not realizing that it was this option causing this.

derkork, 2022-08-23:
Now that option is properly buried so nobody can ever find it :D Great that you found it out. So i guess there is not a lot I can do about this from the app itself, if the OS is intercepting the events. If you agree, I would close this one.

WillAdams, 2022-08-23:
The thing is, there's something about the target definition in the apps which makes this possible --- I can drag to make connections using other node editing programs --- Nodezator for instance, and also Ryven if memory serves.

If you could just keep this in mind for a bit when working on other things/researching Godot --- unfortunately, there's not much which I can find, and "flicks" conflates w/ "flickering" when searching.

https://www.reddit.com/r/godot/comments/r7hi2a/stylus_input/

derkork, 2022-08-23:
I don't have the flick option for my tablet the dialog only has the first "Pen options" tab for me, the other two tabs ("Flicks" and "Touch") are missing.

I checked the documentation for how the flick stuff actually works under the hood:

> The flick detection routine is implemented in the pen input subsystem. Flick detection begins at pen-down and continues until either:
> the sequence of packets received is determined not to be a flick or
> pen-up occurs.
>
> While flick detection is occurring, pen packets are held back and not sent to the system. This must be done because sending  packets may interfere with the flick action that is performed. For example, sending packets during a flick that maps to a copy  action would dismiss what was selected, meaning there would be nothing to copy by the time the action was sent.
>
> As the packets flow into the pen input subsystem, the flick detection routine computes metrics on the length, velocity, time, > and curvature of the motion being performed. With each packet that arrives, the detection routine updates each of these    metrics. As soon as any of the metrics falls outside what would constitute a flick, flick detection ends and the packets are sent through.
>
> Where flicks are detected
> Flick gestures are made possible by the fact that drags are typically performed rather slowly. The user must first target the > start point of the drag, perform the drag and then target the end point. Normally this will take too long to qualify as a flick. However, on inking surfaces quick strokes that would qualify as flicks happen frequently; crossing a 't' is a common example. Thus, by default, flick detection is turned off over inking surfaces and turned on system-wide.

I think Windows will not be able to detect Godot's controls as "Inking Areas" simply because they are no real native controls. They are just a bunch of textures rendered on an OpenGL canvas.

WillAdams, 2022-08-24:
Oh, I see.

Thanks for the deep dive!

closed 2022-08-24 by WillAdams