derkork/openscad-graph-editor #9: Failing at the first hurdle

KeithSloan, 2022-08-18:
Manual says
"You can download the latest release of the OpenSCAD Graph Editor from [GitHub](https://github.com/derkork/openscad-graph-editor/releases) extract it to a convenient location and run the program. No installation procedure is required."

I am on a Mac and have cloned the repro and have a directory openscad-graphic-editor but don't see which if any program to run.
Looking at files with x (execute permission) did not enlighten me.

Tried cloning in /Applications directory with no success.

A post on the OpenScad discussion mailing list seemed to indicate Mac OS is supported.

derkork, 2022-08-18:
Hey Keith, you will need to download the build for OSX from the release page [direct link here for your convenience](https://github.com/derkork/openscad-graph-editor/releases/download/0.2.0/MacOSX.zip), no need to clone the repo.

[image]

If you are on a Mac there is an additional hoop to jump through as Apple doesn't like apps that are not notarized. See [this FAQ entry](https://github.com/derkork/openscad-graph-editor/blob/master/manual/faq.md#i-cannot-open-the-app-on-the-mac-it-says-it-is-damaged).

Let me know if that works for you. Enjoy!

closed 2022-08-18 by derkork

derkork, 2022-08-18:
I have added some clarification to the manual on what to download, so the next person will not encounter this hurdle.