derkork/openscad-graph-editor #23: Add support for OpenSCAD Customizer

WillAdams, 2022-09-10:
The OpenSCAD Customizer:

https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer

is a standardized way of parameterizing an OpenSCAD project.

It would be great if OSGE supported this by:

 - having UI elements for variable creation/assignment which match those which are available to the Customizer
 - instantiated them in the OpenSCAD file in such a way that the equivalent Customizer interface would be created

derkork, 2022-09-15:
We could add some tab where you can create customization variables (i think this can completely replace the variables button):

[image]

There you can add variables put them into categories (which render as Tabs in the customizer) and specify the type and possible values. Then OSGE would render a section on the top of the OpenSCAD file which has the appropriate code for making this work in the customizer. E.g. the two examples shown would render like this:

```openscad
/* [Global] */
...
/* [Flags] */
// Check this if you want rounded edges.
var rounded_edges = false; 

// How should the flabulator operate?
var flabulator_mode = -1; // [1:Horizontal, -1:Vertical]

```
This would work similar for the other supported types (sliders, vectors, etc.). Variables added there would appear in the project tree and could be dragged out like before.

WillAdams, 2022-09-15:
I'd like to see the same kind of graphical widget possibilities as the Customizer is able to instantiate, so:

 - drop-down menus
 - sliders
 - spinners (the up/down arrows)

checkboxes and textboxes seem to already be handled.

A further nicety would be some organizational mechanism which maps to tabs.

I think this should be optional --- but if it is implemented and used in a file it should have an equivalent interface in OSGE as in OpenSCAD itself.

WillAdams, 2022-10-02:
A further consideration here is that modules are written out before variables, so won't be seen by the Customizer --- would it be possible to put variables first?

Having some sort of control for putting some variables below modules would be great --- this is the usual way to hide variables from the customizer.

derkork, 2022-10-04:
I have started some work on this in the [customizer](https://github.com/derkork/openscad-graph-editor/tree/feature/customizer) branch. As a first step I have introduced typed variables (before they are all of type `any`) . For every variable you will be able to specify if you want to have it in the customizer and the editor will place them accordingly (before or or after the first module, inserting a dummy module if necessary). Then depending on the variable type you will have additional options to control the presentation of the variable in the customizer (e.g. the tab in which it should appear, the control that should be used and any range constraints). Finally there will be a section where you can set all customizer variables's values at a glance without having to drag in a lot of  _Set <variable>_  nodes.

derkork, 2022-10-12:
Sneak-Peek:

[image]

derkork, 2022-11-01:
Another sneak peek - customizer is now built-in. Got still quite a few bugs to kill... 
[image]

derkork, 2022-11-07:
It's now released in 0.5.0. This was quite a bit more work than expected. I hope I didn't break something. Let me know how it works for you.

closed 2022-11-07 by derkork

WillAdams, 2022-11-09:
Thanks! I'm testing this out now --- it seems very easy to get the Customizer stuff to happen in OpenSCAD, which is _awesome_ --- but there doesn't seem to be a way to get the same interface in OSGE, which is fine if I'm not missing something.

For example, a construct I use a lot is a drop-down menu:

Units = 1; // [1:Metric, 25.4:Imperial]

[image]

and I'm not seeing a way in OSGE to enforce selecting only 1 or 25.4 (or better still to pick Metric or Imperial).

derkork, 2022-11-09:
You can set this up using the _Fixed Values_ constraint for a variable of type `number`:

[image]

This will give you a dropdown in the new _Variables_ panel:

[image]

It will also render as dropdown in the customizer:

[image]

WillAdams, 2022-11-10:
I'd thought I'd done that, but obviously got something wrong. Thanks! I'll try again.

WillAdams, 2022-11-13:
My apologies --- to confirm that I understand:

 - it is possible to set up support for a customizer in OSGE so that it will show up when exported to OpenSCAD --- I've gotten that working, though I find some of the terminology confusing (I'll work on writing something up on that)
 - Customization support is _not_ included in OSGE's interface --- for the Unit example above, it will be necessary to manually manage inputting 1 or 25.4 and there won't be a matching drop-down in OSGE --- the Customization settings don't change the appearance of the variable handling in OSGE:

[image]

[image]

 
I switch back-and-forth a lot, and it would be nice if there was, but I'll manage w/o if adding that isn't on the table.

Thanks!

WillAdams, 2022-11-20:
It's working great, except one thing --- variables showing up in alphabetical order is a bit awkward --- an option to drag them into a desired order would be awesome. But, of course, they can just be edited when exported to OpenSCAD, so not an essential thing.

derkork, 2022-11-21:
I noticed this as well, but haven't yet found a nice way of incorporating this into the UI. If you have an idea, I'm all ears. :)

WillAdams, 2022-11-21:
The cool thing to do would be to have little textured bar areas at the top left of the area for each variable which would allow dragging them up/down and re-ordering them thusly.

I don't think folks will be making _so_ many customizer variables that that would be too onerous.