derkork/openscad-graph-editor #27: Possible to add a widget which allows controlling order of operations?

WillAdams, 2022-11-24:
In the context of:

https://github.com/WillAdams/gcodepreview

I am finding that I need to ensure that certain operations occur in a certain order --- trying to figure out a work-around, but nothing is presenting itself --- would a new widget be needed? Or is there something in the UI which I am missing?

derkork, 2022-11-24:
I'm not sure i have a firm grasp on the issue right now. Could you make an example that could help me understand better?

WillAdams, 2022-11-24:
Here is an example file:

    include <C:/Users/willa/OneDrive/Documents/RapCAD/Libraries/gcodepreview.scad>

    stocklength = 125;
    stockwidth = 75;
    stockthickness = 6.35;
    toolradius = 1.5875;
    zeroheight = "Bottom"; // [Top, Bottom]
    stockorigin = "Center"; // [Lower-Left, Center-Left, Top-Left, Center]

    generategcode = true;
    retractheight = 3;
    feedrate = 850;
    plungerate = 425;

    //add stockthickness if zeroheight = Bottom
    safetyheight = retractheight; // + stockthickness;

    difference() {
      setupcut(stocklength, stockwidth, stockthickness, zeroheight, stockorigin);

      toolchange(102);
      startspindle (17000);
  
      writecomment("PREPOSITION FOR RAPID PLUNGE");
      rapid(0,0,safetyheight * 2,0,0,safetyheight);

      plungegcutretract(0, 0, 0, 0, 0, -stockthickness, 102, retractheight, plungerate, feedrate); 

    }
    
    closecut();

The command for toolchange must come before the one to start the spindle, then the comment and the rapid command, &c. and finally the closecut command must be the last one --- would it be appropriate to expect a strictly spatial arrangement to work for this sort of thing?

I tried again, and the spatial arrangement seems to be working, so will keep experimenting on that basis.

WillAdams, 2022-11-25:
Until I managed an arrangement which didn't work spatially:

[image]

The order of operations will vary and is dependent on which was first connected --- current is:

    toolchange(tool_number = tool);
    writecomment(comment = "PREPOSITION FOR RAPID PLUNGE");
    rapid(bx = 0, by = 0, bz = (retractheight * 2), ex = 0, ey = 0, ez = retractheight);
    startspindle(speed = spindlespeed);

But if I disconnect everything and change the order things are connected:

    startspindle(speed = spindlespeed);
    rapid(bx = 0, by = 0, bz = (retractheight * 2), ex = 0, ey = 0, ez = retractheight);
    writecomment(comment = "PREPOSITION FOR RAPID PLUNGE");
    toolchange(tool_number = tool);

Even though everything is in the same position.

[flatten.graph.tres.zip](https://github.com/derkork/openscad-graph-editor/files/10088003/flatten.graph.tres.zip)

derkork, 2022-11-25:
I see, this is indeed a bug, the spatial arrangement works only at the top level but not in nested nodes. I'll make a fix for this.

derkork, 2022-11-25:
I also just noticed that you can import a file but there is no way to change that import afterwards. Fixing this as well.

WillAdams, 2022-11-25:
Thanks!

I was just deleting and then re-importing, but yes, being able to modify would be nicer.

derkork, 2022-12-02:
I have now fixed the reimporting, it works a lot better now:

https://user-images.githubusercontent.com/327257/205252887-6b7834e7-6529-4c92-becb-57d7839178c7.mp4