derkork/openscad-graph-editor #28: Click or right-click on a connection line to have an option to disconnect?

WillAdams, 2022-11-27:
Tedious to disconnect multiple inputs until one hits the right one --- could it be possible to click on a connection line to select it and then hit the delete key to disconnect it? Or right-click and then have an option to disconnect that line?

derkork, 2022-11-27:
I noticed this as well. It will probably not be quite easy to implement as this functionality is provided by the underlying graph edit control and this doesn't seem to have a way of selecting connections. I will have to dig a bit and see if I can find a way in there.

derkork, 2022-12-05:
I think I have found an approach into this, but it basically requires me to do some custom connection drawing based on what is in the game engine source code. It looks like it is doable but it's a bit finicky, so it may take a few more days to materialize. But it sure will be a nice improvement.

derkork, 2022-12-06:
Sneak-Peek: 

https://user-images.githubusercontent.com/327257/205858677-74f559ad-0bee-47cf-a00b-89357dad43da.mp4

It still doesn't work very well on some connections and when you zoom it's currently totally off, but we're getting there.

derkork, 2022-12-07:
It seems to be working now. You can right-click any connection and it will be deleted. 

This also enabled another feature to be implemented which I wanted for quite some time which is inserting a reroute node on an existing connection. To do this, do a shift+right-click. This really makes it a lot quicker to layout a graph:

https://user-images.githubusercontent.com/327257/206125672-993e6834-b6db-48c9-ad69-7bc5157b0ce9.mp4

closed 2022-12-07 by derkork