derkork/openscad-graph-editor #30: Variable is created twice

WillAdams, 2022-12-11:
In the attached file:

[box.graph.tres.zip](https://github.com/derkork/openscad-graph-editor/files/10201439/box.graph.tres.zip)

[image]

The variable jo is created twice, lines 38 and 47:

[image]

WillAdams, 2022-12-11:
Here are the libraries as they currently exist:
[Libraries.zip](https://github.com/derkork/openscad-graph-editor/files/10202324/Libraries.zip)

I'm going to try re-factoring things so that the gcp_joinery module doesn't use any variables which aren't properly passed in, which I believe to be the root cause of this, so maybe this isn't a bug, but rather the program not handling an error on the part of the user in a graceful fashion.

derkork, 2022-12-12:
I'm not sure what the correct solution would be here.  There is a variable `jo` defined which has a default value of 0. Later, that variable is assigned a value:

[image]

That in itself is a perfectly valid thing to do, depending on your use case. It doesn't work for your use case because the change to `jo` isn't visible in the called module (OpenScad's variables are really peculiar in that regard) - it is only visible to code directly in the `union` block.  

I checked the documentation if there is a solution for this and there is a thing called [special variables](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Special_variables), which allows you to start a variable with a `$`. Changes to such variables _are_ visible in called modules, so for your use case using something like `$jo` would work. 

I just pushed 0.6.1 which allows variable names to start with a `$` unless they clash with a built-in variable. This should enable your use case to work. 

In regards to the "handling an error on the part of the user", I've got no idea how to handle this because the same setup would be valid in other cases and the program has no way of knowing the user's intent to make a decision if the setup is correct or not.

WillAdams, 2022-12-12:
Thanks!

The whole variable/not variable thing in OpenSCAD has always been very confusing to me, and I'm a bit confused by the variable being defined twice when it only shows once in the graph --- I'll see what I can do to wrap my brain around this.

closed 2022-12-12 by WillAdams

derkork, 2022-12-12:
You could also try some decorator module which would temporarily set the variable without polluting global state:

```openscad
module joint_factor(factor) {
   $jo = factor;
   children()M
}
```

and then 

```openscad
joint_factor(2) setup_stock(...) ...
```

This way the calling code doesn't need to know about global variables and the state is immediately reset after the `joint_factor` module is called.