derkork/openscad-graph-editor #33: Sign / Notarize the App for OSX

derkork, 2022-12-21:
It would be helpful to sign / notarize the app for OSX, so users don't have to take [extra steps](https://github.com/derkork/openscad-graph-editor/blob/master/manual/faq.md#i-cannot-open-the-app-on-the-mac-it-says-it-is-damaged) to make it work there.

- ▶ Figure out how signing/notarization work with Godot and OSX.
- ▶ Figure out how this can be achieved for free - after all the editor produces no income so paying Apple yearly for a developer  account is not an option.
- ▶ Figure out how to integrate the signing process into GitHub actions.
- ▶ Create a signed release.

QuantuMatrix, 2023-02-26:
Epic work my guy, let me know if I can help you get it verified for Apple.

derkork, 2023-04-05:
Fixed in 0.9.3

closed 2023-04-05 by derkork