derkork/openscad-graph-editor #38: Dragging variables with a stylus creates choice menu in the pallet area

WillAdams, 2022-12-24:
If you drag a variable into the editing area using a Wacom stylus in Windows the menu appears in the Variables pallet:

[image]

rather than in the editing area as it does when a mouse or trackpad is used:

[image]

derkork, 2023-02-24:
Hmm i checked this with my Intuous Pro but couldn't reproduce it locally.  Do you have any special settings for the pen? This is how I set up mine:

[image]

WillAdams, 2023-02-27:
I'm having this problem on a Samsung Galaxy Book 12 tablet running Windows Build 1703 --- there aren't many settings/options:

[image]

[image]

[image]

[image]

[image]

[image]

(had to disable Flicks as discussed previously)

[image]

[image]

If no one else is having this difficulty it's probably not worth pursuing --- it doesn't happen on the Wacom One on my MacBook.

closed 2023-02-27 by WillAdams