derkork/openscad-graph-editor #39: Add //!OpenSCAD comment at beginning of file and put modules at the top

WillAdams, 2022-12-25:
Other tools add the line:

    //!OpenSCAD

and it seems to be more workable to have imported modules at the top of the file before variable declarations:

    //!OpenSCAD

    include <C:/Users/willa/OneDrive/Documents/RapCAD/Libraries/gcodepreview.scad>
    include <C:/Users/willa/OneDrive/Documents/RapCAD/Libraries/gcp_joinery.scad>
    
    generategcode = false;

Seems to work better than the current system --- is a viable approach/option?

derkork, 2022-12-26:
Should be doable. Maybe we could do a "preamble" setting, so you can add arbitrary text to be added before the generated code and this can be set up per project as needed.

WillAdams, 2022-12-26:
A further consideration here --- should a semi-colon be included at the end of include?

OpenSCAD doesn't mind, but RapCAD does:

[image]

commit 0c40aa77 (2023-01-04): feat: various improvements and fixes (fixes #39, #40)
closed 2023-01-04 by derkork