derkork/openscad-graph-editor #43: delete key doesn't delete, need to fn+delete (Mac)

philsnow, 2023-03-01:
**Describe the bug**
The delete key (top right of MBP keyboard) doesn't delete nodes.  fn+delete actually deletes nodes.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project, create any node
2. Select the node
3. Press delete key

**Expected behavior**
Node is deleted.

**Desktop (please complete the following information):**
 - OSGE: version 0.8.0
 - OS: MacOS Ventura 13.1
 - hardware: 14" 2021 Macbook Pro, M1 Max
 - I do have Karabiner-elements installed (14.11.0), though without any pre-existing rules that touch the delete key

**Additional context**

When I press the delete key, Karabiner-EventViewer.app shows the keycode `delete_or_backspace`.  When I press fn+delete, it shows the keycode `delete_forward`.

As a workaround, I added the following custom rule to Karabiner-Elements that changes the delete key to send `delete_forward` keycode if the foreground app is OSGE, and now it works as expected.

``` js
{
  "title": "OpenSCAD Graph Editor.app tweaks",
  "rules": [
    {
      "description": "backspace to forward_delete",
      "manipulators": [
        {
          "type": "basic",
          "from": {
            "key_code": "delete_or_backspace"
          },
          "conditions": [
            {
              "type": "frontmost_application_if",
              "bundle_identifiers": [
                "de.janthomae.openscadgrapheditor"
              ]
            }
          ],
          "to": [
            {
              "repeat": true,
              "key_code": "delete_forward"
            }
          ]
        }
      ]
    }
  ]
}
```

philsnow, 2023-03-01:
Actually this workaround doesn't work great, because now I can't easily use the delete key to backspace in text entry fields in OSGE.  I'm removing the custom rule and just using fn+delete to delete nodes for now.

derkork, 2023-03-02:
The "Delete" Key on a MBP is actually a backspace key. In previous iterations of MBP it was labeled "⌫"  while newer iterations label it (somewhat misleadingly) "delete".  Compare that to a full magic keyboard there is actually a real "delete" key:

[image]

I will add an additional keybind that allows to delete nodes also with the "backspace" key (which is labelled "delete" on the MPB).

commit 81e332e7 (2023-03-02): feat: allow to delete nodes with backspace key

fixes #43
closed 2023-03-02 by derkork