derkork/openscad-graph-editor #47: Visual drawing of polygon

derkork, 2023-03-26:
**Is your feature request related to a problem? Please describe.**
Whenever I want to create a polygon I need to imagine how it would look like. In addition the handling of 2D point arrays isn't really great either, so I cannot easily remove points or insert one in the middle.

**Describe the solution you'd like**
It would be great if  there was a special node which allowed you to draw a polygon and edit the points with the mouse, also inserting and deleting points and also typing in the position of a point if one needs to be exact.  Snapping to a grid would be a bonus. Another bonus would be if a piece of additional data could be given to each point (e.g. a rounding factor when using polyround). The node would then simply create an array of vector2 or vector3 objects based on the drawing.

**Describe alternatives you've considered**
Alternatively the nodes for creating a vector2/vector3 array should get features for inserting or reordering items. Drawing a polygon would still help visualizing stuff.

WillAdams, 2023-03-31:
There is an interactive tool for drawing polygons which has been posted to the OpenSCAD mailing list (or at least that's what my uncertain organic memory leads me to believe) --- perhaps a workable example to look at?

derkork, 2023-03-31:
I could only find a discontinued tool named "Polygon Rabbit" and the mailing list does no longer seem to be mirrored on the web, do you happen to know a name or a link?

derkork, 2023-03-31:
And i found this one: https://climberg.de/polygoneditor/

WillAdams, 2023-03-31:
I think it was:

https://github.com/butcherg/wxpolygon/releases/tag/v1.2

(which I found on the /r/openscad subreddit --- mental note, mirror all interesting things from the mailing list to that subreddit)

derkork, 2023-04-07:
Actually I came to the realization that this would be a lot more useful if one could add measurements to the drawing, ideally from the variables. This would be really great for parameterization of the sketch. FreeCad can do such a thing, but  to replicate this in OSGE I would need to have at least some minimal sketch solver, which may take a while to get functioning.

derkork, 2023-04-08:
Yes i also use functions extensively for this purpose, helps to keep the graphs clean. The solvespace library is sort of undocumented, so I'm not sure how far I could get with it. I'd probably also prefer something written in C# as to not add native dependencies which would make distribution across platforms more problematic. I think this goes to the backburner for a little while until I have some attack plan for this.