derkork/openscad-graph-editor #50: App crashes on mac even after running command to remove xattr

origamifreak2, 2023-04-05:
**Describe the bug**
App crashes on mac even after running `sudo xattr -rd com.apple.quarantine OpenSCAD\ Graph\ Editor.app`. I also tried removing all xattr with this command `sudo xattr -c OpenSCAD\ Graph\ Editor.app` based on [this code](https://stackoverflow.com/a/4833168), but still no luck. The error it gives is "The application “OpenSCAD Graph Editor” can’t be opened."

**To Reproduce**
Steps to reproduce the behavior:
1. Download app
2. run command `sudo xattr -rd com.apple.quarantine OpenSCAD\ Graph\ Editor.app` or `sudo xattr -c OpenSCAD\ Graph\ Editor.app` on the app
3. try to open the app
4. App crashes (see crash log in Additional context)

**Expected behavior**
App should not crash

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macOS Monterey Version 12.6.3 (21G419)
 - Hardware: MacBook Pro (16-inch, 2019)
    Processor 2.6 GHz 6-Core Intel Core i7
    Graphics AMD Radeon Pro 5300M 4 GB

**Additional context**
[crash_log_2023-04-04.txt](https://github.com/derkork/openscad-graph-editor/files/11160501/crash_log_2023-04-04.txt)

derkork, 2023-04-05:
Running the command to remove the quarantine should no longer be necessary. Which version did you download?

derkork, 2023-04-05:
[image] [Tesrt](https://trello.com/c/1IVclzPr/1-tesrt)

origamifreak2, 2023-04-05:
I was using 0.9.2. I just noticed that you released 0.9.3 with apple notarization and it opens just fine on my mac.

origamifreak2, 2023-04-05:
Updated my op to include the app version number

derkork, 2023-04-05:
Ok great, I was really afraid i messed something up with the notarization. I fought the last 48 hours with it until it finally worked on my Mac and if the notarized version didn't work for you I would have been really out of ideas 😌 

So I guess I can close this now, as the fix is to upgrade to the notarized version. Thank you very much for the quick reply.

closed 2023-04-05 by derkork