derkork/openscad-graph-editor #62: Allow duplicating a node with all connections

WillAdams, 2023-09-16:
Often, when working on a project, one will need an almost duplicate of a node --- most of the connections will be the same, but one or two will need to be different.

Being able to duplicate the node and maintaining all the connections would make that much more efficient, since it wouldn't be necessary to reconnect the connections which are in common betwixt the twain.

derkork, 2023-09-18:
So the idea would be you press "duplicate" and it duplicates all selected nodes, offsets them and then adds additional connections to the duplicates, e.g. this:

[image]

would become this:

[image]

WillAdams, 2023-09-18:
Yes, just that --- but usually the node in question is a module.

WillAdams, 2023-09-19:
If it can be done, I'd really appreciate it, and would find it a big help in more complex projects.

Thanks for considering it!

derkork, 2023-09-19:
Yes I think it would be a great addition.

closed 2023-09-21 by derkork

derkork, 2023-09-21:
Will be in release 0.11.0, as soon as the build runs through. Apple is doing another set of shenanigans with their licenses so currently the build aborts when trying to sign the build for OSX. I hope that this gets resolved soon.

derkork, 2023-09-21:
There we go: https://github.com/derkork/openscad-graph-editor/releases/tag/0.11.0

derkork, 2023-09-21:
Ah I forgot, it's Ctrl+Shift+D / Cmd+Shift+D if you want the extra connections.

commit 2961ca64 (2023-09-21): feat: allow to keep incoming/outgoing connections when duplicating nodes

fixes #62