derkork/openscad-graph-editor #63: Easier/clearer branching

WillAdams, 2023-10-16:
It can be difficult to do more complex problems which require multiple if/then checks.

For example, I worked up a series of checks which worked:

[image]

but going back to it later when re-writing/drawing/creating this, it didn't seem clear at all, and the arrangement of the nodes/lines wasn't communicating clearly to me.

After a bit of experimentation, I arrived at:

[image]

which is all true, and I believe works out to the same logically, but some sort of CASE SELECT or other branching option would be welcome.

Just realized that there is an & option --- but didn't see that in the manual --- so maybe this is just a request that & be added to the documentation --- going to try again using it.

derkork, 2023-10-16:
I have added a paragraph about the boolean operators to the manual. One option to clean this up a little bit would be some string enum node where you can plug in a string and have a range of outputs which are set to true if the input string matches one of the options:

[image]

This could reduce the 10 comparison nodes to two in your case. You'll still need the boolean operators though as this is the logic for selecting the outputs, but it would at least reduce the visual clutter a bit and using strings as enums is a pretty common use case so I have a feeling this would be a node worth adding.

WillAdams, 2023-10-16:
Didn't find it at first, since I was expecting it to be under Branching.

Unfortunately, "And" and "Or" are difficult to search for --- maybe put && in the text to make it findable w/ a Search?

derkork, 2023-10-17:
I have added && and || to the text so its easier to find. Good catch!

derkork, 2023-10-31:
So we now have this: 

[image]

Which renders to:

```openscad
if (((kind == "cube") || (kind == "cuboid"))) {
    cube();
}

if ((kind == "sphere")) {
    sphere();
}

```

commit 11c18e12 (2023-10-31): feat: add string enumeration test node

fixes #63
closed 2023-10-31 by derkork