derkork/openscad-graph-editor #66: concat doesn't work as expected

WillAdams, 2023-10-29:
If one has a concat node:

[image]

it makes the code:

        opengcodefile(fn = concat(Base_filename, ".nc"));
        opendxffile(fn = concat(Base_filename, ".dxf"));
        opensvgfile(fn = concat(Base_filename, ".svg"));

which doesn't work as expected, since the text which is thus assembled is rendered as:

    ECHO: ["export", "nc"]

Naturally wrapping in str is the fix:

    fn = concat(str(Base_filename, ".nc"));

and this can be done in the interface:

[image]

but I was surprised that it was necessary --- is it strictly necessary that the OpenSCAD code be thus represented? Does one ever concatenate things other than strings?

"concat" doesn't appear in the manual AFAICT, so maybe this is just another documentation request?

derkork, 2023-10-29:
`concat` is for concatenating arrays, it's a built-in function of OpenSCAD (https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#concat). For concatenating strings you can use `str` directly (https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/String_Functions#str). It also allows you to specify multiple inputs. 

[image]

WillAdams, 2023-10-29:
Ah! That would explain it. Thanks!

closed 2023-10-30 by derkork