derkork/openscad-graph-editor #67: Possible to add an export of a graph/selection as an SVG?

WillAdams, 2023-11-02:
Currently, in order to use a Graph in documentation it is necessary to make a screen grab, which results in a pixel image, which is size-specific --- would it be possible to have a menu entry which would export either the entire graph, or just the current selection as an SVG or PDF?

That would allow for didactic usage in traditional publications, or pretty much anything.

derkork, 2023-11-03:
I don't think this is something that can be done in a reasonable amount of time. The graph is rendered by the underlying engine which is pixel based so there is no vector image to begin with. It would have to be generated completely from scratch using the available graph data. While this is surely possible to do, it's not exactly trivial and neither is rendering to SVG or PDF.  Given the limited amount of development time I have available, and also assuming that this is probably not something a lot of people would need I think the effort for implementing this beyond what I would like to spend on it.

WillAdams, 2023-11-03:
Understandable. It was a thought I had, and I was hoping the architecture would be like to Display PostScript on the NeXT and allow this easily.

It looks like the position is encoded for each node in the file?

    Position = Vector2( 0, 18 )

    Position = Vector2( 750, 0 )

If push comes to shove I'll try to find time to dig into this.