derkork/openscad-graph-editor #68: Possible to easily dim lines leading to disabled nodes?

WillAdams, 2023-11-18:
When re-working a module, one thing I've found myself doing regularly was disabling most of the nodes:

[image]

but while this helps the 3D preview in OpenCAD, it doesn't make the node/editing view that much more explicable.

If the lines leading to disabled nodes were similarly dimmed, it would be easier to pick out which calculations were being used and which weren't.

derkork, 2023-11-20:
Would this work?

[image]

closed 2023-11-20 by derkork

WillAdams, 2023-11-20:
That looks about perfect! Thanks!

derkork, 2023-11-21:
Thanks. Just released 0.13.0 which has this.

commit 0a9f7360 (2023-11-20): feat: dim connections from/to disabled nodes

fixes #68