derkork/openscad-graph-editor #71: Functions can be hard to pull out and visually are a block

WillAdams, 2024-03-08:
Would it be possible to add an option for adding an icon/pictogram to user-defined functions?

derkork, 2024-03-09:
This should be totally doable. I could add a selection of icons from which you can pick and maybe also allow to give a color in which the node is tinted, so this would then look similar to the built-in icons: 

[image]
[image]

Would this work for you or did you have something different in mind?

WillAdams, 2024-03-09:
A selection of text characters/symbols would be cool, but I was thinking more about allowing the user to create their own icon which would be expressive of what the function does.

GraphSCAD had a bunch for its various custom functions, some of which worked well, while others were a bit inscrutable.

The challenging thing here of course is allowing the user reasonable creativity while guiding them into a result which will look well with the balance of the interface.

Probably it will be necessary to allow a textual label in addition to the graphical representation, which seems redundant, but folks accept labeling of files and so forth.

derkork, 2024-03-11:
Something like this could do it. I still need to figure out where to store the custom icons, ideally they would be embedded in the `.graph.tres` file so projects are easily portable across machines. For now I think it's enough to support square PNGs the editor can auto-scale it to the required resolution, non-square formats would be scaled to fit while keeping aspect and filled with transparent  letterboxes/pillarboxes.

[image]

WillAdams, 2024-03-11:
Ah, hadn't considered the file | embedding / missing resource aspect.

Square makes sense, and having an available list to pick from would be a great thing, even if folks only used them as placeholders.

One potential source would be:

https://thenounproject.com/

thawkins, 2024-03-13:
Extending this later to key into a function library that people could contribute to, package the icon, function definition and description.

derkork, 2024-03-13:
Yes, funny enough the same thought crossed my mind yesterday when I realized that the editor can consume text based libraries but right now has no way to make or consume libraries made  directly in the editor. Of couse you can  still export them as SCAD and consume them this way but the icons will not survive an export to SCAD (unless I do something ridiculous like embedding them base64-encoded into the exported file which will make it utterly unreadable). I'll put that on the list. Thanks!

derkork, 2024-03-25:
So embedding works but i'm not sure about the rendering.. The icon kinda gets in the way with the function parameters.. 
[image]

WillAdams, 2024-03-25:
Create room at the top for the icon? Increase the overall width?

derkork, 2024-03-25:
On top seems to work quite well. 
[image]

derkork, 2024-03-25:
I think i'll keep the name in. It looks cleaner without it but just from the icon it's probably hard to detect what the module/function was: 
[image]