derkork/openscad-graph-editor #72: Sample *.graph.tres file(s) to get started

eMPee584, 2024-09-18:
This app introduces a new file format and it would be nice to find a few sample files included. That said, the [manual](https://github.com/derkork/openscad-graph-editor/blob/master/manual/manual.md) seems well made.. :+1: :eyes: 
it's not yet integrated in the application itself, is it? :thinking:

derkork, 2024-09-21:
I'm not quite sure what to make of this. The graph.tres format is internal to the application and it's not intended to be read/written by other applications. The underlying format is Godot's TRES format which is a subset of [TSCN](https://docs.godotengine.org/en/stable/contributing/development/file_formats/tscn.html). Is there a specific thing you want to do with the files?

eMPee584, 2024-09-24:
Well, it's just nice to be able to explore a program's functionality from a sample file instead of creating one from scratch, that was the motivation to file an issue.

derkork, 2024-09-26:
Ah thank you very much for the clarification. That makes a lot of sense. I'll see if I can come up with some examples to try.